berndpfrommer/tagslam #3: Input data format

XiaoxingChen, 2019-05-27:
Hi Bernd,
    I'm going to use AprilTag as groundtruth in my project to test my VIO and I'm trying to use "tagslam"  to get a tag map. However I cannot run my own ros bag on tagslam with the command mentioned in README: 
1. The tagslam.launch example in README subscribe detected tags' topics while the code itself actually read the rosbag. Maybe the README might be too old to be a reference? 
2. Then I tried to create a folder with the same structure as example_1. I know what to write into camera_poses.yaml and cameras.yaml, but the tagslam.yaml is quite confusing,  could you please release some explanations about the meaning of the key words in tagslam.yaml? For example, what does the "rig" mean, as well as "perch".

Thanks!

berndpfrommer, 2019-05-27:
Yes, the Readme is outdated. I just finished a complete rewrite of the
code. Will update documentation this week.
I'm traveling right now, will respond in more detail later today.

berndpfrommer, 2019-05-28:
OK, here some more explanations on how this 'bodies' thing is supposed to work (this will be documented cleanly within the next few weeks). Since you are shooting for ground truthing of VIO, I will use a cut-down version of tagslam.yaml from  example_12, because that is actually from a VIO related project.

```

body_defaults:
  position_noise: 0.05
  rotation_noise: 0.01
default_body: top_floor
bodies:
 - top_floor:
     is_static: true
     ignore_tags: [166, 213, 223, 237, 367, 381, 392, 354, 405, 585]
     default_tag_size: 0.150
     max_hamming_distance: 2
     pose:
       position:
         x: 0
         y: 0
         z: 0
       rotation:
         x: 0
         y: 0
         z: 0
       position_noise:
         x: 0.0001
         y: 0.0001
         z: 0.0001
       rotation_noise:
         x: 1.0e-05
         y: 1.0e-05
         z: 1.0e-05
     type: simple
     tags:
       - id: 6
         size: 0.15000000
         pose:
           position:
             x: 1.92018
             y: 0.5987
             z: 2.2390
           rotation:
             x: 0.0
             y: -1.570796327
             z: 0.0
           position_noise:
             x: 0.0001
             y: 0.0001
             z: 0.0001
           rotation_noise:
             x: 1.0e-5
             y: 1.0e-5
             z: 1.0e-5
 - rig:
     ignore_tags: [381]
     is_static: false
     type: camera_rig
     odom_topic: "/fla/vio/odom"
     odom_frame_id: odom
     odom_acceleration: 1.0
     odom_angular_acceleration: 1.0
     T_body_odom:
       position:
         x:  0.02568531
         y:  0.0238219
         z:  0.026066910
       rotation:
         x:  -7.66219656e-04
         y:  2.84301166e-02
         z:  3.14080874

```
Forget about body_defaults. That's only used when you don't specify the noise parameters of a bodies pose. So let's get to the bodies. Here are the rules:
Every tag has to be associated with a body, and every tag is assumed to have the same relative position with respect to that body. You cannot have a "free tag". It must belong to a body. If a tag is found in the wild, it will be attached to the default body. In this case, the default body is "top_floor". Because we were actually mapping the top floor of the building. So the building itself is a body, and the tags are attached to it. We know the pose of this body with respect to the world, so we give it here. As you can see from the noise parameter (which translates to a pose prior in the optimizer), the top_floor body is pretty much bolted to the world frame system, and is identical with it. The parameter "is_static" indicates: this body does not have a time dependent pose, so the pose graph optimizer will maintain a single SE3 object for this body,.
The body type is "simple". That pretty much says it all. At the moment the only other relevant type is "board", which means: auto-generate and attach a set of tags. Not important if you don't use apriltag boards, and just use simple individual tags.

For some tags we had measured the position with a laser pointer. For those, we specify their pose, with some noise parameter. For example tag 6 was measured, so we put it's pose into tagslam.yaml. NOTE: you must obviously measure at least one (1!) tag's pose, or just set it to the origin. If none of the tag's poses are given, there is no way to ever localize.

Now that the "world" body (top_floor) has been covered, let's move on to the "rig". That's another body, but it does not have tags attached. Rather, it has cameras attached to it (the cameras must specify that body in the cameras.yaml!). You also see that it's not a static body, i.e. the rig's pose is time dependent and must be computed from scratch for every frame. You must specify the pose of at least one camera with respect to the rig, which is actually done in camera_poses.yaml. This is necessary to localize the rig via this chain when seeing e.g tag 6:
top_floor_pose(known)->tag_6_pose(known)->tag_observation->camera_world_pose->rig_pose(via known camera-to-rig-pose).
Now if you are running some other odometry for e.g. the rig body (in our case VIO), you can feed that in via the odom_topic as shown in the input file. The odom must have the same frequency as your camera images, and will be used to initialize and constrain the rig pose before applying the tag observations. If your VIO is not in the camera frame (it is usually in the IMU frame), you must specify the transform between rig body and odom reference frame, see above.

Hope this helps to some extent. I will close this issue once the documentation is in place.

berndpfrommer, 2019-06-19:
There is now a TagSLAM documentation page:

https://berndpfrommer.github.io/tagslam_web/

closed 2019-06-19 by berndpfrommer