berndpfrommer/tagslam #5: set_parameters error when applying stereo images in sync_and_detect.launch

hwscut, 2019-09-25:
I try to use sync_and_detect.launch in my stereo camera, so I modify the image_topics parameter.
But get the error like this.
I want to know how to solve this problem

changes in sync_and_detect.launch 
```
<rosparam param="image_topics"> ["/mynteye/left/image_raw", "/mynteye/right/image_raw"]</rosparam>
<rosparam param="image_output_topics"> ["/annotated_images/left/compressed", "/annotated_images/right/compressed"] </rosparam>
<rosparam param="tag_topics"> ["/detector/left/tags", "/detector/right/tags"]</rosparam>
```

run:
```
 roslaunch tagslam sync_and_detect.launch
```

get error:
```
[ERROR] [1569418228.264122670]: Tried to advertise a service that is already advertised in this node [/sync_and_detect/compressed/set_parameters]
```

berndpfrommer, 2019-09-25:
I cannot reproduce this error in my setup.
Please send me your complete sync_and_detect.launch file, and a small rosbag (just a few images) to bernd.pfrommer@gmail.com

hwscut, 2019-09-26:
this is my complete sync_and_detect.launch file.

```
<launch>
  <arg name="output" default="screen"/>
  <arg name="bag" default=""/>
  <arg name="detector_type" default="Mit"/>
  <arg name="start_time" default="0.0"/>
  <arg name="duration" default="-1.0"/>
  <arg name="images_are_compressed" default="true"/>
  <arg name="annotate_images" default="true"/>
  <!-- for list of valid tag families, see apriltag_detector.h
       0 = 36h11
       1 = 25h9
       2 = 16h5
  -->
  <arg name="tag_family" default="0"/>
  <node pkg="tagslam" type="sync_and_detect_node" name="sync_and_detect"
    output="$(arg output)" clear_params="True">
    <rosparam param="image_topics">
    ["/mynteye/left/image_raw", "/mynteye/right/image_raw"]
    <!--["/pg_17274483/image_raw/compressed"]-->
    </rosparam>
    <rosparam param="image_output_topics">
    ["/annotated_images/left/compressed", "/annotated_images/right/compressed"]
    <!--["/annotated_images/compressed"]-->
    </rosparam>
    <rosparam param="tag_topics"> ["/detector/left/tags", "/detector/right/tags"]</rosparam>
    <!--<rosparam param="tag_topics"> ["/detector/tags"]</rosparam>-->
    <!-- To pass through odometry messages, add them like so:
	<rosparam param="odometry_topics"> ["/vio/odom"]</rosparam>
    -->
    <param name="bag_file" value="$(arg bag)"/>
    <param name="detector_type" value="$(arg detector_type)"/>
    <param name="tag_family" value="$(arg tag_family)"/>
    <param name="black_border_width" value="1"/>
    <param name="annotate_images" value="$(arg annotate_images)"/>
    <param name="images_are_compressed" value="$(arg images_are_compressed)"/>
    <param name="start_time" value="$(arg start_time)"/>
    <param name="duration" value="$(arg duration)"/>
    <param name="output_bag_file" value="$(arg bag)_output.bag"/>
    <param name="max_number_frames" value="1000000"/>
  </node>
</launch>
```

I don't run any bag, just type:
```
roslaunch tagslam sync_and_detect.launch
```
when setting just one image_topics, it can run correctly. 
But setting two image_topics, errors come.
so I don't know it is  my parameter error or  a bug in code

berndpfrommer, 2019-09-26:
OK, without bag, I get the same error as you.
This is some weird error related to ROS image transports. I never encountered it because I never tested with multiple cameras online. It was primarily developed for processing bags.

I have seen the error before in other contexts, but actually despite the error, the image transport worked fine. Aside from the error, what else is not working?

Bear in mind that for sync_and_detect to work, your left and right images must have identical header.stamp time stamps.

I'm trying to fix the error, but can you try in the meanwhile if anything else doesn't work?

berndpfrommer, 2019-09-26:
Can you try this as well: change the code as follows in sync_and_detect.cpp, and recompile with 'catkin build'?
This will disable using the compressed transport (I think I forced this because I was playing back data from bags, and had to force the use of a compressed image transport).
 
SyncAndDetect::View::View(
    ImageTransport *it, const std::string &topic,
    ImageOdometrySync* sync) : topic_(topic), sync_(sync) {
    //image_transport::TransportHints th("compressed");                                                                                                                                                            
    //sub_ = it->subscribe(topic, 10, &View::callback, this, th);                                                                                                                                                  
    sub_ = it->subscribe(topic, 10, &View::callback, this);
  }

hwscut, 2019-09-27:
I have tested the codes above, it works fine.

another issue i want to make sure:
Notice that the codes cann't publish odometry message.
so if i want to publish the synchronous odometry message,  how to add codes.
just add  "pubs_.push_back(nh_.advertise<nav_msgs::Odometry>(topic, 10))" in   SyncAndDetect::subscribe() ?
```
  void
  SyncAndDetect::subscribe() {
    std::vector<std::vector<std::string>> tpv;
    tpv.push_back(imageTopics_);
    tpv.push_back(odometryTopics_);
    imageTransport_.reset(new image_transport::ImageTransport(nh_));
    sync_.reset(new flex_sync::Sync<Image, Odometry>(
                  tpv, std::bind(&SyncAndDetect::processImages,
                                 this, std::placeholders::_1,
                                 std::placeholders::_2)));
    for (const auto &i: irange(0ul, imageTopics_.size())) {
      views_.emplace_back(new View(imageTransport_.get(),
                                   imageTopics_[i], sync_.get()));
      pubs_.push_back(nh_.advertise<apriltag_msgs::ApriltagArrayStamped>(
                        tagTopics_[i], 10));
    }
    // warning: never tested odom code!
    for (const auto &topic: odometryTopics_) {
      odoms_.emplace_back(new Odom(nh_, topic, sync_.get()));
      pubs_.push_back(nh_.advertise<nav_msgs::Odometry>(topic, 10));// like this ?
    }
  }
```

berndpfrommer, 2019-09-27:
Sorry, but I don't quite understand what you want to accomplish.
Odometry messages are usually published by a separate ROS node, for instance a VIO or other SLAM code. Why would you want to republish the subscribed messages?
In fact, usually there is no need to feed the odometry messages into sync-and-detect. I don't even know why I put it in there. Would've been easier to do the detection first, and then merge the bag with the tag detections and the odometry by hand.
Along these lines a warning: your odometry messages must have the same header.stamp as your image messages. There is no "approximate sync" available right now. You would have to write a separate code, probably using the ROS approx sync tools, to force the time stamps to be identical, the odom can be used for TagSLAM.
The only advantage for using sync_and_detect is that if sync_and_detect does not spit out any messages, you know your time stamps don't line up.

Oh, and another note: I never used the online odometry code, see code snippet below. I put it there for completeness, but had no good test case. I did test it when running sync_and_detect on bags. Whatever, I recommend not using it, and feeding the odometry messages directly into tagSLAM. Should work if your time stamps are synced.
```
// warning: never tested odom code!                                                                                                                                                                            
    for (const auto &topic: odometryTopics_) {
      odoms_.emplace_back(new Odom(nh_, topic, sync_.get()));
    }

```

hwscut, 2019-09-28:
ok thanks, I use tagslam with odometry of other VIO ros node, it can work.
I just wonder the odometry codes you put in sync_and _detect file.
perhaps, it is easier to  generate the bag with vio and tag detection, like the bag in example.

closed 2019-09-29 by hwscut