berndpfrommer/tagslam #6: Different black boarder size

AndreasVaage, 2019-09-29:
Hi, I notice, that the tags have to be unique and of the same family to be used, but can they have different with of the black boarder? We have a single 6x6 apriltag grid calibration board where the is a double outer black boarder of each tag (The default generated from kalibr). We want to use this board with other tags with a black boarder of only 1 with. Is this supported?

AndreasVaage, 2019-09-29:
Sorry, I now see you specifically stated the number of borderbits has to be the same. For the future, apriltag 3 does not longer seems to natively support the option of [adjusting the number of borderbits](https://github.com/AprilRobotics/apriltag/issues/45).

closed 2019-09-29 by AndreasVaage

berndpfrommer, 2019-09-29:
Andreas, thanks for pointing this out.

I'd love to use the Apriltag3 version as well, but it's not conveniently accessible in the ROS apriltag package I'm using right now.

Chao (@versatran01), what are your plans for integrating Apriltag3 into your package? Obviously we'd still want the option to switch to the MIT detector so we can handle border width of 2. Any comments?

versatran01, 2019-09-29:
If you think it is stable then I can work of master branch to replace old umich with the new one.
If you can do a PR then it's even better