berndpfrommer/tagslam #7: Prbloem with running tagslam on my own bags

WaldumA, 2019-10-08:
Hello, and thank you for this repository. :)
I would like to use this repository in order to find the trajectory of a camera. It works really well on all the examples, but when I try to run it on my own bags it seems to skip the entire processing of bag step. The output is the following:

`NODES
  /
    tagslam (tagslam/tagslam_node)

ROS_MASTER_URI=http://localhost:11311

process[tagslam-1]: started with pid [10781]
[ INFO] [1570554675.755388265]: optimizer mode: slow
[ INFO] [1570554675.756214627]: added body grasp_lab with 1 tags
[ INFO] [1570554675.756251801]: added body rig with 0 tags
[ INFO] [1570554675.758748015]: found 1 cameras
[ INFO] [1570554675.759699400]: camera cam0 has known pose!
[ INFO] [1570554675.759809502]: no distance measurements found!
[ INFO] [1570554675.759854734]: no coordinate measurements found!
[ INFO] [1570554675.759892180]: no plane measurements found!
[ INFO] [1570554677.352922721]: reading from bag: /home/waldum/id25.bag_output.bag
[ INFO] [1570554677.357002454]: number of tag   topics: 1
[ INFO] [1570554677.357222968]: number of image topics: 0
[ INFO] [1570554677.357397941]: number of odom  topics: 0
[ INFO] [1570554677.364303754]: done processing bag, total wall time: 0.000333957
[ INFO] [1570554677.364581648]: updater performance:

[ INFO] [1570554677.365035348]: final error: 0
writeMeasurementDiagnostics = tot:         2us per: 2+-0(2-2) count: 1 
finalOptimization           = tot:       284us per: 284+-0(284-284) count: 1 
writeTimeDiagnostics        = tot:       291us per: 291+-0(291-291) count: 1 
writeTagDiagnostics         = tot:       310us per: 310+-0(310-310) count: 1 
writeErrorMap               = tot:       500us per: 500+-0(500-500) count: 1 `

I ran sync_and_detect.launch on my original bag, and then feed the output bag into tagslam.launch. I tried to echo /detector/tags to see if there were any errors in the topic, but the output looks just like the tag topics in the examples. 

I have set the tag with id=25 as the one known tag in origo, and initiated the camera one meter above the tag in tagslam.yaml and camera_pose.yaml. I also set the default tag_size according to my apriltags. As far as I understand all other found tags will automatically be set to the default body so I do not have to specify them, as long as they are in the same family. Before running the launch file I also set all the intrinsic camera parameters in cameras.yaml according to my calibration.

I am rather confused since I can not see any difference between my input and the examples. Do you have any suggestions to where my mistake is? 

Best regards, Ambjørn Grimsrud Waldum

WaldumA, 2019-10-08:
I was able to locate the error, I had written the wrong tagtopic in cameras.yaml. /detector/tag instead of /detector/tags. My bad, everything works now! :)

closed 2019-10-08 by WaldumA

berndpfrommer, 2019-10-08:
From the fact that tagslam finishes within microseconds, it's almost certainly an issue with ROS topics not matching. The topics in cameras.yaml need to match the ones in your bag.

Can you please post your tagslam.yaml, cameras.yaml, and run rosbag info on the bag that you are feeding into tagslam, i.e. the one you created with sync-and-detect?

berndpfrommer, 2019-10-08:
You beat me by 3 seconds :)