berndpfrommer/tagslam #8: cannot get continuous trajectory in /tagslam/odom/body_rig

hwscut, 2019-10-09:
try to run my own bag
use msckf_vio  to generate continuous vio trajcetory
but I cannot get continuous trajectory in /tagslam/odom/body_rig
I want to know whether this situation is correct?
note these codes:
```
  void TagSlam::publishBodyOdom(const ros::Time &t) {
    for (const auto body_idx: irange(0ul, nonstaticBodies_.size())) {
      const auto body = nonstaticBodies_[body_idx];
      Transform pose;
      if (graph_utils::get_optimized_pose(*graph_, t, *body, &pose)) {  // here is condition
        auto msg = make_odom(t, fixedFrame_, body->getOdomFrameId(), pose);
        odomPub_[body_idx].publish(msg);
        if (writeToBag_) {
          outBag_.write<nav_msgs::Odometry>(
            "/tagslam/odom/body_" + body->getName(), t, msg);
        }
      }
    }
  }
```

so can I get continuous trajectory of tagslam/odom/body_rig??

berndpfrommer, 2019-10-09:
Combining msckf_vio odom with tag observations should definitely work, and give you a reasonably continuous trajectory (subject to possible jumps in the odometry, and jumps on loop closure, but both of these can be worked on).
Can you post more information on what's going wrong? First of all: is your mskf_vio working? Can you post an rviz graph of the msckf odom? Is that trajectory smooth?
Once that works: are there any areas where you can run tagslam without the odometry (maybe in places where there are plenty of tags)? Maybe just run it on a single tag? That will give you a very jittery trajectory, with occasional jumps, but at least you can tell if basic parameters like intrinsic calibration etc are correct.

When you combine odometry and tags, suppress all tags except for one (which must have a defined pose in the world). Does the trajectory look smooth then? If not, where does it jump?

Other things worth posting so I can give you better feedback:
- some camera images showing the tags (so I can tell the quality of the images)
- tagslam.yaml and cameras.yaml
- rviz screen shot of where the trajectory jumps, showing odometry and tag poses (enable "TF")

berndpfrommer, 2019-10-10:
A couple of clarifications and hints
1) I just tested for the first time to run sync_and_detect live, with odometry as well, and it works. Except that it couldn't handle the case of raw (uncompressed) images. I have fixed that and just pushed it. Please do 'git pull origin master' to update.
2) The syncing actually happens at the camera: that's when the header.stamp is set. So long as left and right camera have identical header.stamp, everything should work: the time stamps should flow through msckf_vio, arrive at sync_and_detect, and have identical header.stamp. The sync_and_detect then runs the tag detector, and produces the detected tags, again with matching header.stamp. Sync_and_detect does not re-publishes the odom messages, it just filters the images based on it, i.e. it only runs the tag detector when it has a left images, a right image, and an odom message, all with the same time stamps. Only then does it publish the tags.
3) If things don't work, carefully step through all your ros nodes, and run "rosnode info" on every one of them. Do they they subscribe to valid topics? Topic mismatch is the most common problem.

Here is the relevant snippet from my sync-and-detect.launch
```
  <arg name="images_are_compressed" default="false"/>
  <arg name="annotate_images" default="false"/>
  <!-- for list of valid tag families, see apriltag_detector.h
       0 = 36h11
       1 = 25h9
       2 = 16h5
  -->
  <arg name="tag_family" default="0"/>
  <node pkg="tagslam" type="sync_and_detect_node" name="sync_and_detect"
    output="$(arg output)" clear_params="True">
    <rosparam param="image_topics">
      ["/ovc/cam_0/image_raw",
       "/ovc/cam_1/image_raw"]
    </rosparam>
    <rosparam param="image_output_topics">
      ["/annotated_images/cam_0/compressed",
       "/annotated_images/cam_1/compressed" ]
    </rosparam>
    <rosparam param="odometry_topics">
      ["/vio/odom" ]
    </rosparam>
    <rosparam param="tag_topics">
      ["/ovc/cam_0/tags", "/ovc/cam_1/tags"]
    </rosparam>

```

hwscut, 2019-10-11:
I think my topics  are correct.

## 1.by typing below commands, I cannot get continuous trajectory.
```
roslaunch msckf_vio our.launch
roslaunch tagslam sync_and_detect.launch
roslaunch tagslam tagslam.launch config_dir:=./ run_online:=true
```

## 2.But if I type likes below operation, it works.
```
roslaunch msckf_vio our.launch
roslaunch tagslam sync_and_detect.launch
```

After running sync_and_detect.launch, I can get  _output.bag in .ros.
```
rosbag play _output.bag
roslaunch tagslam tagslam.launch config_dir:=./ run_online:=true
```
This time, it can get continuous trajectory.
so I really confuse about this situation.

berndpfrommer, 2019-10-11:
How does the non-continuous trajectory look like? Does rviz show any odometry? Does the tagslam node publish any odometry?

Can you run rosbag info on the "_output.bag" and post the output? That way I can see what topics are in there, and maybe guess which ones are missing in the other case (if this even is the problem).

The only other guess I have is that it has to do with time stamps. I usually do "rosbag play --clock foo.bag" when playing back from a bag. Also, note that in this case you should use "rosparam set use_sim_time true", which you must set *before* you run rviz (and any other ros node that relies on clock). In the scenario 1) above, do you also play from bag, or is that data then directly coming from the camera?

Can you collect a bag with the original data (collect the "compressed" image topic), just maybe a minute or so, and send it to me (bernd.pfrommer@gmail.com), along with the cameras.yaml, camera_poses.yaml and tagslam.yaml?

berndpfrommer, 2019-10-18:
Is this still an issue?

hwscut, 2019-10-18:
I have sent the data to your email, just a moment ago. hope that you can fix it

berndpfrommer, 2019-10-19:
I had intense difficulty getting the odometry to work right, in the end I got it kinda working, and I did not see the jumps in the tagslam output that you mentioned.
The first thing to check for jumps is the raw odometry itself. There are occasionally jumps in the msckf_vio, when it loses features, because that is when it is compounding all the visual feature tracks into the state update.
If the odometry looks good, then the only reason for jumps should be when tagslam tries unsuccessfully to close the loop. When the gap is too big you will get subgraph error warnings, and a jump in trajectory. But that did not seem to be the problem in this case.

hwscut, 2019-10-20:
I finally find the problem. 
When I play the bag slowly, it can get continuous trajectory.
```
rosbag play -r 0.8 my.bag
```

But when playing the bag in normal speed, the trajectory is non-continuous. 
I check the odometry from vio and the /tagslam/odom_raw/body_rig.
It shows the odometry from vio is continuous, but /tagslam/odom_raw/body_rig is non-continuous, which causes the final error result.

so I think whether the computing consume of  tagslam  is large , which causes that it cannot get the odometry from vio timely. 
When running online, tag detection, vio and tagslam are running simultaneously.
If we get the _output.bag firstly,  and run tagslam later,  it can work well, because only tagslam is running without tag detection and vio generation.

closed 2019-10-23 by hwscut