berndpfrommer/tagslam #9: run two-circle mapping error

hwscut, 2019-10-14:
I capture data with 2 circle of a scene rather than only 1circle.
because I think it can generate loop_closure  for better mapping accuracy.

But when running the second circle, the trajectory of /tagslam/odom  flies away
so it is reasonable?

berndpfrommer, 2019-10-14:
Sorry, I really can't say why the trajectory "flies away". Most likely it's a misdetected tag.
Yes, unless you have good scene overlap during the mapping run, i.e. always multiple tags are visible at the same time, the accuracy of the map depends on the VIO. You need to place more tags if you want to reduce the importance of the VIO accuracy.
Please send me your tagslam.yaml, cameras.yaml, camera_poses.yaml, and the bags with the tag detections and the VIO odometry messages. These files are usually not that big, and will allow me to understand the problem better.

hwscut, 2019-10-14:
I have sent test data to your email(bernd.pfrommer@gmail.com). please verify my data.
And compared to one tag, multiple tags can significantly improve the accuracy of tag poses.
For why? just because more corners or other optimization reason?

berndpfrommer, 2019-10-14:
Can you verify your T_body_odom transform?
Right now you have:
```
    T_body_odom:
       position:
         x:  0.002295070911334022
         y:  -0.04575619724634699
         z: -0.02592267864649928
       rotation:
         x:  7.11107524e-04
         y: -5.30136450e-03
         z: -1.57119987e+00

```
But with that transform, the cameras look crooked on the robot:
[image]
Are they really attached like that?
Looks like you are running VINS (never used it, but heard good things about it). Now VINS is using a reference body frame. Often, for VIO the reference frame is actually the IMU. From your tagslam.yaml file, I can see you have cam 0 (left mynteye) at the identity pose, meaning your tagsam body frame is the left camera. In this case T_body_odom is the transform that takes coordinates *from* the IMU frame *to* the left camera frame. Can you double check that you are indeed using this transform?

I will try to look more at your data later today or tomorro.w

hwscut, 2019-10-14:
in my calib.yaml 
```
cam0:
  T_cam_imu: [-0.00041244197749279454, 0.9999925925770441, -0.0038268371609805787,
    0.04575619724634699, -0.9999956479498335, -0.0004012578698114888, 0.0029228536593719177,
    0.002295070911334022, 0.002921296460031294, 0.003828026013936469, 0.9999884060547046,
    -0.02592267864649928, 0.0, 0.0, 0.0, 1.0]
```
I capture data by hand-hold, so the init cam pose would be arbitrary.

berndpfrommer, 2019-10-15:
The issues you are seeing are typical problem with loop closure: your odometry drifts for a good while, and eventually on loop closure, the optimizer gets in trouble when it tries to distribute all this accumulated error across the whole path.
In situations like this, it helps to switch to the "slow" mode, and do a full graph optimization, e.g. every 500 frames. Once you have a good location of the tags (copy from the poses.yaml to tagslam.yaml), you can switch back to "fast" mode.

Also, during mapping runs, you need to set the noise on your odometry artifically low, like so
```
    odom_translation_noise: 0.005
     odom_rotation_noise: 0.001
```
Otherwise, if a tag is not seen properly on the first view, it tends to propagate that error to later frames, and that error may throw off e.g. your orientation: all of a sudden the trajectory climbs or rises out of the plane.
The "fix" is that for a mapping run, you artificially reduce the noise on your odometry. Once you have an idea of where the tags are, you can specify their poses (give them a large position error, i.e. 5m).

At that point, you may also want to use all the information you have: if a tag is flat against the wall, and upright, specify the rotational degree of freedom! I usually just look at the angles obtained from the first mapping, and correct them to be e.g. sqrt(1/2)*pi etc. Measuring the z-coordinate of a tag is also easy, so specify it in the tagslam.yaml, and give x and y the approximate values you obtained from the mapping run (with large uncertainty). That way your trajectory is less likely to take off.

Lastly, I'd advice you to go the loop counter clock wise, rather than accumulating all this error until you finally see the tags.
[image]
You can see that there is still some weird crap happening near the starting point, where the loop closure occurs. But the raw VIO also looks screwy there, so not sure what happened at that moment.

I think you can get good results with this data, you need to just help it along a little bit. For instance, you can take a cheap laser pointer to measure down that hall way and get the distance of the furthest tag, then maybe one that is 1/2 way down the corridor.

hwscut, 2019-10-17:
ok  thanks for your advices.
But as I know, the general slam can generate more perfect pose estimation of camera after loop closure. And can get an accurate map.
In tagslam, it can only generate map which depends on vio. If my vio drifts, the tag poses in the map  would have error. 
Can I  use the trajectory after loop closure to generate the tag poses, I think it would be more accurate.  This is for why I have to run two-laps in a scene.

berndpfrommer, 2019-10-17:
slow mode: run incremental optimization, but calculate the error, and, if error jumps dramatically, run a full optimization
fast mode: close eyes, run only incremental optimization, and hope all went well
Running in slow mode is safer, but in reality, in the vast majority of cases you get away with fast mode. Only use slow when fast bombs out.

TagSLAM always uses the *entire* data set available at the time you call the rosservice /tagslam/dump. The poses that are written to ~/.ros/poses.yaml are the result of optimizing across *both* loops, i.e. they will corrected by the loop closure.

Having that said, it is another question to what extend the optimizer will distribute the mismatch detected upon loop closure back to the tags. Chances are it'll stretch some of the VIO deltas, and barely move the tags, which are far away at that point. Would be interesting to see what happens if you played the data *backwards*, i.e from the first loop closure in reverse to the start. I'm pretty certain the results would be different. But it's hard to set the reverse playing up, because you'd have to change all the time stamps etc.

What I want to say is that my confidence in the optimizer doing wonders for your tag poses upon loop closure are fairly low. I would imagine it just stretches the nearby VIO deltas a bit to fit the data.

hwscut, 2019-10-17:
thanks for your explanation, I will explore the codes later

closed 2019-10-17 by hwscut