berndpfrommer/tagslam #10: Tags with doubble border bits

WaldumA, 2019-10-21:
Hi!
I have been running your algorithm on paper tags with single border bits, and it has worked really well. However last week I did some underwater tests where I used a aluminium board and some laminated papers, both with double border bits. When I tried launching synch_and_detect.launch on the recorded bag it was not able to detect any tags at all. 

My first thought was that the underwater images probably was distorted in such a way that the tags were no longer recognizable by tagslam even though they looked quiet clear to me. So I tried to record some bags of the aluminium board and papers in air with a good camera. On these bags synch_and_detect.launch are able to detect one or two tags in a few of the frames, but it should have been able to detect a lot more than 1-2 in every frame. 

From the tagslam wiki: 'All tags must have the same number of border bits'. Based on this I assumed that it would be fine to use double border bits as long as all tags had a double border. 

Are double border bits supported, and if that is the case is there any parameters I have to specify that I am using double border bits? 

All of the tags I have used are of the 36h11 family. 

Best regards,
Ambjørn

berndpfrommer, 2019-10-21:
You need to set the apriltag detector correspondingly in sync_and_detect.launch. There is a rosparam for that:

  `<param name="black_border_width" value="2"/>
`
I suspect the few detections you got are accidental. Please try running with different border width and let me know if that fixes the problem.

WaldumA, 2019-10-21:
Yes it solved the issue, thank you :)

closed 2019-10-21 by WaldumA