berndpfrommer/tagslam #12: distance_measurements unused

SeanGuoXuan, 2019-12-06:
Consider this case:

- Bodies:
     - body0:
          - pose(...)
          - tag0
               - pose(...)
          - tag1
              - pose(...)
     - body1:
          - tag2
              - pose(...)
- distance_measurements:
     - tag0-1:
          - tag1: 0
          - tag2: 1
      ...
     - tag0-2:
          - tag1: 0
          - tag2: 2
      ...

Body0 is defined as default body has prior pose, body1 doesn't. Body0 has two tags(0, 1) attached, body1 has one tag(2) attached. Define 2 distance measurements: 
    (1) from tag0 to tag1; 
    (2) from tag0 to tag2; 
If you run tagslam, then you will end up with distance_measurements "tag0-2" **unused**. Tracing the source code I found: at initial stage, body0, tag0, tag1, tag2 are marked as optimized but not body1.  It causes measurement "tag0-2" failed to be added to optimizer. See the code below:

void Measurements::tryAddToOptimizer() {
      for (const auto &v: vertexes_) {
        if (graph_->isOptimizableFactor(v) && !graph_->isOptimized(v)) {
          auto fp = std::dynamic_pointer_cast<factor::Factor>((*graph_)[v]);
          fp->addToOptimizer(graph_.get());
         }
       }
 }

Is this a bug or designed with purpose?  I think measurement config is very powerful tool to improve  system accuracy. Tag measurement should not rely on the body status (optimized or not) that tag attached to.

berndpfrommer, 2019-12-06:
In the scenario you describe, the tag distance measurements now become time dependent, because body 1 is not static. In other words: the tag distance can change over time, so any measurement would need to carry a time stamp that marks the time at which the distance was measured. This is currently not supported.
To support it we'd have to change/expand the input file format to include a time stamp for the measurement, but also check when new tag observations are provided by the camera, if the current time stamp is in sync with the distance measurement, and then enter it into the graph at that point.
Bear in mind that just adding a measurement into the graph at a random time t (unsynchronized) will make things very tricky, because for that time, the pose of body 1 may be ill determined, as just a tag distance measurement is not enough to constrain the body 1 pose at time t completely.

SeanGuoXuan, 2019-12-07:
Sorry I didn't explain the problem clearly. Body 1 is static by design. In my tagslam.yaml, I do specify "is_static: true" for body 1. The only difference between body 0 and 1 is that body 0 has prior pose but body 1 not. I expect body 1 pose can be optimized once camera observe appearance of tag 1 and 2. 

For your extended solution of adding timestamp to the measurement may not be practical. It requires a lot extra configuration effort. I think there maybe two possible solutions:

1. add pose guess to body 1 with high noise variance. So the measurement "tag0-2" can be added to optimized list in graph class. Let the system figure out(? not sure my understanding is correct) appropriate pose later on.  This is bypass the restriction for adding factor to optimizer( "graph_->isOptimizableFactor(v)").

2. Add this factor to full graph or sub graph rather than optimized graph. Then there is a chance factor can go into optimized graph if subgraph error falls below some threshold. If this make sense, can the author provide hints how to implement this?

Also, I still don't understand the logic behind "Graph::isOptimizableFactor". Before adding the factor to optimized_ map, it loops through all vertex connected to("getConnected(v)"),  anything is not optimized then return not optimizable. But "Body:body1_0" is not a factor by my understanding.

berndpfrommer, 2019-12-07:
OK, I understand now what you want to accomplish.

Your option 1.) is the easiest, and that's what I'd do. You need to set the pose approximately right for the first run, then grab the optimized pose from the output ~/.ros/poses.yaml and run again.
As a variant to this: do the first run *without* the measurement, let tagslam figure out body pose 1, then put that as pose prior, then enter the measurement.

I had initially planned to have just static measurements, and it seemed to me that in such cases, you'd be able to provide a pose prior for the static bodies. Being able to insert the measurement factors at the beginning of the run, statically, simplifies the code considerably, and sidesteps a range of potentially thorny issues that come about when you insert the measurements later (your option 2).

Even when doing it statically at the beginning, care must be taken such that the optimizer is not asked to solve an underdetermined system. That's what isOptimizableFactor() is about: before it enters a measurement, it checks that all values connected to this measurement are already pinned down by other factors. In your case, one of the tags is *not* pinned down (because body 1 pose is not given). Without this check, it would add the measurement right away, along with the tag pose, and then the optimizer would bomb out because a distance measurement is not enough to determine the full location, let alone the orientation of the tag.

If you add the factor right away to the full graph, you will probably run into issues because now the measurement factor may connect parts of the graph, and the subgraph discovery will follow those edges, finding (erroneously) that body poses can be determined which actually cannot, because the measurement factor is not sufficiently powerful to establish full pose. This can all fixed by some checks to be sure, but will need some careful testing/debugging.

Lastly, there is option 3): apply the remaining measurement factors at the end of the run, again. This will be slightly less powerful than adding them right away when the body pose is established, because adding the measurement earlier may force the map to be correct earlier, and may so prevent bomb-outs that happen when the map is bad. Unless the early measurement application prevents bomb-out, the two approaches should lead identical results.

I will try to implement the fix 2.), and if that leads to tricky code, resort to option 3.)

berndpfrommer, 2019-12-07:
I guess I should have looked at the code before replying. Here are the facts:
- what is already implemented is option "2", i.e. measurements are used at the earliest possible moment.
- the messages about "unused" are in fact a *bug*. The measurements *are*  used, and you should be able to find them in the error map under ~/.ros/
- it turns out test 13 actually already covers that scenario, and also suffers from the bogus messages.

I fixed the bug and pushed the changes to the master (not yet root). It should not affect your results, just the printout.

SeanGuoXuan, 2019-12-08:
Cool. 
Also, I notice if measurement is problematic, then tagslam terminated with warning "Check all your input noise settings!". This makes sense but it left camera_poses.yaml empty file somthing like:
> cam0:

Tagslam finalize program without proper camera poses:
```
  void TagSlam::finalize(bool optimize) {
    doDump(optimize);
```
But this is another issue.

closed 2019-12-08 by SeanGuoXuan