berndpfrommer/tagslam #15: irange problem

m4xr1sk, 2020-02-15:
got some problem with irange function while building

`tag_slam.cpp:798:57: error: no matching function for call to ‘irange(long unsigned int, std::vector<boost::shared_ptr<const nav_msgs::Odometry_<std::allocator<void> > > >::size_type)’`

I have installed libboost-all-dev version (1.67.0.1+b1)

do you have any suggestion?

berndpfrommer, 2020-02-15:
Now that's an interesting one.

I looked at the source, and the header file is included etc. Besides it seems to build fine for me.

Curiously, the relevant line for me is 797, not 798, so we may be slightly out of sync with the source code, but I assume this is the offending line:
```
for (const auto odomIdx: irange(0ul, odomMsgs.size())) {
}
```

One reason why this could fail is that odomMsgs.size() is not of type 0ul (unsigned long int). Usually, size_t is of type unsigned long, but on your system, it may not.

Could you please post what OS and hardware you are trying to compile on?
As a work around, please try "0u" or "0" instead of "0ul" at the relevant line above.

m4xr1sk, 2020-02-15:
I'm on a Raspberry PI 4
`Linux raspberrypi 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux`

I was out of sync because I've tried to rewrite the for loop without using irange.
I've made my changes in a fork, but now I have other problems with gtsam

m4xr1sk, 2020-02-15:
ok with my fork it builds

closed 2020-02-15 by m4xr1sk