berndpfrommer/tagslam #17: any example with Intel T265

m4xr1sk, 2020-02-15:
I'm using a Raspberry PI 4 with an Intel T265 camera.

How can I use the pose estimated by the camera to navigate where no tags are detected?

berndpfrommer, 2020-02-15:
The fact that you are using a PI makes me nervous because it means that you are probably trying to set up a real-time system with very limited CPU power. This will almost certainly not work out, see below.

Having that said, here are the ingredients necessary to give it a try:
1) A ROS driver that publishes an odometry message from the camera. From the documentation, it looks like this driver might do it: https://github.com/IntelRealSense/realsense-ros
2) Run the apriltag detector node from the tagslam repo and have it subscribe to the images from the T265. Detecting the tags will probably be the most CPU time consuming step. You may have to run really low frame rate, maybe down to 1Hz.
3) Have TagSLAM use the "odometry" feature (see examples in the tagslam_test repo)

The problem is that TagSLAM expects synchronized input, which means it will *drop* any odometry messages unless there is a matching detection from *both* cameras as well. If you don't have plenty of CPU power, either the left camera, or the right camera will drop the image because the node can't keep up with detecting. But in that case, TagSLAM will not process anything, because all 3 messages (left, right cam, and odom) need to be present to set up the pose graph.

To solve this problem, one could modify the detector node to publish an empty detection message if it cannot keep up with the processing. This is currently not implemented, and may not be that simple (you will likely have to go multi-threaded etc).

Even if that succeeds, it's not all roses: TagSLAM does not prune the graph or does fancy key frame detection. Over time, the graph gets big, and TagSLAM slows down. On a desktop machine, I can run for 15mins and still be almost real-time at 20fps. This will not be the case for a Pi. If on the other hand you first do a mapping run to get the location of all tags, then you can enter their poses into the tagslam.yaml config file and for subsequent runs use the "amnesia" feature, which will only use the last two time steps in the graph. That is very efficient and may run in real-time on a Pi.

m4xr1sk, 2020-02-17:
Thanks for your answer.
I've removed everything from your repo and I've installed the apriltag_ros from AprilRobotics.

With almost everything working it's not so bad:

```
---
header: 
  seq: 15707
  stamp: 
    secs: 1581973837
    nsecs: 297384262
  frame_id: "camera_fisheye1_optical_frame"
detections: 
  - 
    id: [0]
    size: [0.144]
    pose: 
      header: 
        seq: 14926
        stamp: 
          secs: 1581973837
          nsecs: 297384262
        frame_id: "camera_fisheye1_optical_frame"
      pose: 
        pose: 
          position: 
            x: -0.270945467495
            y: 0.106951459896
            z: 1.60038056259
          orientation: 
            x: 0.979833045917
            y: 0.023703238903
            z: 0.105563896602
            w: 0.167992923446
        covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 15708
  stamp: 
    secs: 1581973837
    nsecs: 464058161
  frame_id: "camera_fisheye1_optical_frame"
detections: 
  - 
    id: [0]
    size: [0.144]
    pose: 
      header: 
        seq: 14931
        stamp: 
          secs: 1581973837
          nsecs: 464058161
        frame_id: "camera_fisheye1_optical_frame"
      pose: 
        pose: 
          position: 
            x: -0.270459901915
            y: 0.106866681067
            z: 1.59710081064
          orientation: 
            x: 0.97793140728
            y: 0.0232716658957
            z: 0.116321192368
            w: 0.171982477096
        covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

Between 2 frames it's less than 200ms.
I'm going back to the std apriltag_ros because I'm new to ROS and I'm trying to understand how it works... then I'll try again the tagslam and the "amnesia" feature

berndpfrommer, 2021-01-14:
Example 22 is for a T265.

closed 2021-01-14 by berndpfrommer