berndpfrommer/tagslam #18: Camera pose, rotation conventions

matheusbg8, 2020-06-12:
Hi Bernd,

I am testing your code and I am having issues defining the camera poses, more specifically the rotation. Since the ROS rotations are usually defined as quaternions, rotation matrix, or Euler RPY, my first attempt was to define the camera rotation vector as Euler RPY (XYZ). However, it doesn`t seem the right representation.

After reading your code I found the following:
[image]

Can you please send me a link that can help me to figure out how your rotation representation works? I could not find references in your documentation https://berndpfrommer.github.io/tagslam_web/input_files/

Can I use https://www.andre-gaschler.com/rotationconverter/ to convert RPY to "Axis with angle magnitude (radians) [x, y, z]" ? 

Thanks!

berndpfrommer, 2020-06-14:
Sorry for the late reply!
Yes, you are right, the documentation is lacking. Please leave this issue open until I get around to documenting that the rotation is specified in axis * angle notation, i.e. vee(log(R)).

berndpfrommer, 2020-11-11:
Thanks again for pointing this out!
Fixed with commit 6cbe70d69add6153abb565ffc65ea660fb76aecb of the tagslam_web repo.

closed 2020-11-13 by berndpfrommer