berndpfrommer/tagslam #19: can not get tagslam pose

zzl20, 2020-09-23:
Hi, i used my owndata(rosbag) to run tagslam,  "the subgraph err :" always cout "0", and can not publish odomtry!

berndpfrommer, 2020-09-23:
Can you please provide more information to go by? The beginning of the tagslam console output, your tagslam launch files, your tagslam.yaml, cameras.yaml, camera_poses.yaml, maybe a ``rosbag info <name_of_rosbag>``?

zzl20, 2020-09-24:
> Can you please provide more information to go by? The beginning of the tagslam console output, your tagslam launch files, your tagslam.yaml, cameras.yaml, camera_poses.yaml, maybe a `rosbag info <name_of_rosbag>`?

The rosbag info is:
[image]
The tagslam launch file is :
[image]
The cameras.yaml is :
[image]
The camera_poses.yaml and the tagslam.yaml is same as example file.

The output is :
[image]
[image]
[image]

i want get camera pose, but it is not optimize and can not publish odomtry!

zzl20, 2020-09-24:
> Can you please provide more information to go by? The beginning of the tagslam console output, your tagslam launch files, your tagslam.yaml, cameras.yaml, camera_poses.yaml, maybe a `rosbag info <name_of_rosbag>`?

hi , i have find the problem, thank you very much, you can close this issue

closed 2020-09-24 by zzl20