berndpfrommer/tagslam #21: Wrong package name in package.xml

eduardoiscar, 2020-11-10:
When trying to install dependencies using rosdep, it throws an error as it can not locate   <depend>standard_msgs</depend>

I believe this is an error, as   <depend>std_msgs</depend> is included a couple of lines further down  in the file. 

Removing the mentioned line installed all dependencies succesfully on a fresh ros:melodic-base install  and I am able to run the example without issues

berndpfrommer, 2020-11-10:
Would you mind posting the error message you are getting?
I usually don't use rosdep to install dependencies so I'm not sure exactly what and how to fix.
Thanks!

eduardoiscar, 2020-11-11:
The command I run to install all dependencies is:
`rosdep install --from-paths src --ignore-src --rosdistro melodic -y ` 

The error it throws is:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tagslam: Cannot locate rosdep definition for [standard_msgs]
```

The reason is that the package.xml file of the tagslam project includes the following line that has to be deleted:   <depend>standard_msgs</depend>. 
The correct way to specify that the package depends on standard_msgs is   <depend>std_msgs</depend>
which is actually already present in the package.xml file a couple of lines further down. 

Hope this helps! 
Best, 
Eduardo

berndpfrommer, 2020-11-11:
Now I get it.
Thanks much for pointing this out. This has been fixed with commit 09a6f1fccac689f3e5b32c67fc41073f517cffa7

closed 2020-11-11 by berndpfrommer