berndpfrommer/tagslam #23: online delay with mono camera

zthcool, 2021-02-25:
It runs smoothly since the beginning,but it comes delay after a while.It's image topic is sync, but it's TF changes will delay.

zthcool, 2021-02-25:
The delay comes with the next message
"max count reached, running full optimization!"
at the beginning ,this message will flash to pass,but after about 1min the message will take a lot of time,then the TF in rviz comes delay.

zthcool, 2021-02-25:
I think I just know how to use it. set "amnesia" to true in the tagslam.yaml(the parameter is not in the file in the beginning,you should write it yourself),there will no delay,but you should know all the tags pose in advance.So set "amnesia" to false it will calculate all the tags itself with delay.when you get all tags pose you can run it with real-time.So the amnesia" just like a save/load map switch.
So if there is a function to save all tags calculate in a yaml files would be more convenient.
Thanks for your great work!

berndpfrommer, 2021-02-25:
You can avoid the sudden delays by bumping max_num_incremental_ops to a very large value. But in the long run that won't help you, since the size of the growing graph will eventually catch up with you, even when just performing incremental inference aka iSam. So yes, for this case the crude "amnesia" hack is the way to keep the graph bounded. (The cleaner way would be to marginalize out old variables, but that is somewhat against the idea of tagslam)

Here is how to get all the tag poses:
1) run without amnesia until you have a good map of the environment.
2) perform a "dump": 
```
rosservice call /tagslam/replay
```
This will write a set of files to ~/.ros/. One of them is poses.yaml, which has the tag poses that you can cut-and-paste into your tagslam.yaml.

zthcool, 2021-03-09:
I cannot find the poses.yaml in ~/.ros and I searched all computer there is no poses.yaml file.I only find tag_coners.txt in the ~/.ros

closed 2021-05-12 by berndpfrommer