berndpfrommer/tagslam #26: why I launch the tegslam.launch and the program crashed after load the params?

hjxwhy, 2021-03-28:
I have tested the program on ubuntu 18.04 + gtsam 4.0.0 and ubuntu 16.04 and gtsam 4.0.0, the compile process don't have any errors, but when I roslauch the program, it crashed.

started roslaunch server http://hjx:34805/

SUMMARY
========

CLEAR PARAMETERS
 * /tagslam/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /tagslam/bag_file: /home/hjx/tagslam...
 * /tagslam/bag_start_time: 0.0
 * /tagslam/camera_poses/cam0/pose/R: [1000000.0, 0.0, ...
 * /tagslam/camera_poses/cam0/pose/position/x: 0
 * /tagslam/camera_poses/cam0/pose/position/y: 0
 * /tagslam/camera_poses/cam0/pose/position/z: 0
 * /tagslam/camera_poses/cam0/pose/rotation/x: 0
 * /tagslam/camera_poses/cam0/pose/rotation/y: 0
 * /tagslam/camera_poses/cam0/pose/rotation/z: 0
 * /tagslam/cameras/cam0/camera_model: pinhole
 * /tagslam/cameras/cam0/distortion_coeffs: [-0.161604, 0.093...
 * /tagslam/cameras/cam0/distortion_model: radtan
 * /tagslam/cameras/cam0/intrinsics: [1404.917, 1403.5...
 * /tagslam/cameras/cam0/resolution: [1920, 1200]
 * /tagslam/cameras/cam0/rig_body: rig
 * /tagslam/cameras/cam0/rostopic: /pg_17274483/imag...
 * /tagslam/cameras/cam0/tagtopic: /detector/tags
 * /tagslam/has_compressed_images: False
 * /tagslam/max_number_of_frames: 1000000
 * /tagslam/playback_rate: 10.0
 * /tagslam/tagslam_config/bodies: [{'grasp_lab': {'...
 * /tagslam/tagslam_config/body_defaults/position_noise: 0.05
 * /tagslam/tagslam_config/body_defaults/rotation_noise: 0.01
 * /tagslam/tagslam_config/default_body: grasp_lab
 * /tagslam/tagslam_config/tagslam_parameters/max_num_incremental_opt: 50
 * /tagslam/tagslam_config/tagslam_parameters/max_subgraph_error: 50.0
 * /tagslam/tagslam_config/tagslam_parameters/minimum_viewing_angle: 12.0
 * /tagslam/tagslam_config/tagslam_parameters/optimizer_mode: fast
 * /tagslam/tagslam_config/tagslam_parameters/pixel_noise: 1.0
 * /tagslam/tagslam_config/tagslam_parameters/subgraph_abs_prior_position_noise: 0.1
 * /tagslam/tagslam_config/tagslam_parameters/subgraph_abs_prior_rotation_noise: 0.1
 * /tagslam/tagslam_config/tagslam_parameters/use_approximate_sync: False

NODES
  /
    tagslam (tagslam/tagslam_node)

auto-starting new master
process[master]: started with pid [10137]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d0933d2e-8fae-11eb-92d1-081f712eafca
process[rosout-1]: started with pid [10150]
started core service [/rosout]
process[tagslam-2]: started with pid [10168]
[tagslam-2] process has died [pid 10168, exit code -11, cmd /home/hjx/tagslam_root/devel/lib/tagslam/tagslam_node __name:=tagslam __log:=/home/hjx/.ros/log/d0933d2e-8fae-11eb-92d1-081f712eafca/tagslam-2.log].
log file: /home/hjx/.ros/log/d0933d2e-8fae-11eb-92d1-081f712eafca/tagslam-2*.log

berndpfrommer, 2021-03-28:
Is this on an AMD or Intel processor?
Is it crashing when you run the example?
How can I reproduce the crash?

hjxwhy, 2021-03-28:
> Is this on an AMD or Intel processor? Is it crashing when you run the example? How can I reproduce the crash?
> […](#)

Thanks for your reply!
It's on Intel process.  I just download the code as the readme of tagslam_root and compiled without error.  I run the example as readme guide and crashed. as the terminal output it may be run the tagslam node crashed but i can't find where has wrongs

berndpfrommer, 2021-03-28:
Where did you get your GTSAM library from? DId you install it from the
repository, following the instructions in the tagslam_root repo?
I know there are some issues with GTSAM being compiled with the compiler
flag march=native which causes crashes on certain architectures, but I have
not yet figured out how to address that.
Not sure if that works but you could also try to install the current
development version 4.1.x of GTSAM. The build is currently broken, but
maybe the old binaries will still install. Those should be compiled without
-march=native.

https://launchpad.net/~borglab/+archive/ubuntu/gtsam-develop

sudo apt-add-repository ppa:borglab/gtsam-develop
sudo apt update
sudo apt install libgtsam-dev libgtsam-unstable-dev

berndpfrommer, 2021-03-28:
And could you please run it under the gdb debugger as described here?

http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

That should give you a stack trace so we know a bit better where it's
crashing.

hjxwhy, 2021-03-29:
@berndpfrommer Thanks for your help, I have solved the problem. As you say, the problem is caused by the GTSAM being compiled with the compiler flag march=native, I solved it by adding `cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..` instead of `cmake ..`, building GTSAM from the source. I close this issue.

closed 2021-03-29 by hjxwhy