berndpfrommer/tagslam #27: Is here a bug after removing vertex and adding factor to graph, but don't add factor to optimizer?

hjxwhy, 2021-04-17:
https://github.com/berndpfrommer/tagslam/blob/c0c5c76aeb169be1779b405a99604f57a1471f82/src/graph_utils.cpp#L124

Hi, I have read your code carefully, and I found here when you filtered the vertex that doesn't be optimized you just add the factor to the graph and doesn't add those factor to the optimizer. Is it right?

berndpfrommer, 2021-04-17:
I frankly no longer remember how this logic works. I just remember it's very tricky. Is there a particular case that is broken that you want to fix? It's easier to look at a concrete case.
You are wondering why it is not adding the factor to the optimizer? I'm more puzzled why it is adding the value to the optimizer.

hjxwhy, 2021-04-18:
Yes, I am wondering why it is not adding the factor to the optimizer. The pose values are added to the GTSAM's new value but the factors aren't added to the GTSAM's graph. It is a corner case I have tested many times and the function "https://github.com/berndpfrommer/tagslam/blob/c0c5c76aeb169be1779b405a99604f57a1471f82/src/graph_utils.cpp#L89" has no effect.

slmMuc2fly, 2024-03-06:
Slam anytime

berndpfrommer, 2024-03-30:
closing this as will-not-fix

closed 2024-03-30 by berndpfrommer