berndpfrommer/tagslam #28: When taking odometry to tagslam, the callback do not works

hncsfan1996, 2021-05-08:
I have specify the odometry topic and transformation from odometry to odom and others as the website https://berndpfrommer.github.io/tagslam_web/input_files/ shows. And I am confirmed that the time stamp of odometry and images is identical. But when I run tagslam,  the callback do not works.
Is there any advice for you ? Thank you.

berndpfrommer, 2021-05-08:
Can you please post the console output of tagslam?
Also, do
```
rosnode info tagslam
```
(your tagslam node may have a different name). What topics is tagslam subscribing to? Is data coming on those topics (rostopic echo)?

hncsfan1996, 2021-05-09:
Thank you very much for your reply. I made a .zip about my console output in two situations(image only and image with odom), and also my bag information. Could you please help me to solve it?

抄送: "the ***@***.******@***.***&gt;;

berndpfrommer, 2021-05-09:
Not sure where that zip file of yours went. I don't see it anywhere. If you already want to send me files, send them to bernd dot pfrommer at gmail dot com. Please send all your yaml files, the launch files you are using and a rosbag with the odometry and images. To keep the rosbag size to a minimum record just a couple of seconds. That should be enough to diagnose the problem.

berndpfrommer, 2021-05-12:
The "solution" is to set ``use_approximate_sync: true`` in the ``tagslam.yaml`` file

closed 2021-05-12 by berndpfrommer

hncsfan1996, 2021-05-13:
Thank you so much for taking time to help, I could take odom topic to  tagslam successfully now.