berndpfrommer/tagslam #30: ERROR with "/tagslam/path/body_rig"

ThinkPig826, 2022-05-17:
[image]
I just started using this program, why am I getting the path like this.

berndpfrommer, 2022-05-17:
To properly diagnose this you need to provide more information.
The first thing to look at is the intrinsic camera calibration. How was that one done, what is the reprojection error, was it entered correctly?
Also I can see a calibration board in the picture. Could be that the tag detector occasionally decodes those tags, and that the tag numbers collide with the tags you have on the wall (16, 17).
Lastly, you are quite far away from the tags, and there are only two of them, fairly close together. You won't get a very good pose estimate (although it should be much better than what you are getting).

ThinkPig826, 2022-05-18:
Thank you very much for your reply!
   I increased the number of tags, reduced the distance between the camera and the tag,  the positioning effect was significantly improved.
[image]

In addition, I feel that the system is relatively stuck when running. Do you have a good method?
Finally, thanks again for your reply

berndpfrommer, 2022-05-18:
Not sure what you mean with "relatively stuck".
- if you get no pose updates for a while, it is most likely because no tags are detected. A common reason for that is that no tags are in view (duh) or that the detector fails to detect due to motion blur. Use larger tags, more tags, and *better lighting*.
- if Tagslam bogs down (computes a long time) you can do this: first map out all the tags. Then cut-and-paste the tag poses (look in ~/.ros for the output) into your tagslam.yaml file. Then set "amnesia" to "true" in tagslam.yaml (see instructions). Now tagslam should no longer bog down.

ThinkPig826, 2022-05-19:
ok，thanks！

closed 2022-05-19 by ThinkPig826