berndpfrommer/tagslam #33: Tagslam gives bayes tree node not found when I try to tune the algorithm for the bag file of my setup.

dt1729, 2023-01-23:
Setup:  Markers are placed in a gazebo simulation with turtle bot and odometry is fed from the simul itself. I have given information about two markers in the tagslam.yaml file but with respect to the global coordinate frame of gazebosim. 

It gives me this error that I'm trying to understand: 
``` [ERROR] [1674501247.977188733]: /tagslam: Requested the BayesTree clique for a key that is not in the BayesTree
[tagslam-1] process has died [pid 9562, exit code 255, cmd /home/dt/AddverbWork/slamalgos/tagslam_ws/devel/lib/tagslam/tagslam_node __name:=tagslam __log:=/home/dt/.ros/log/ea0002dc-9b51-11ed-997c-91c581871bdd/tagslam-1.log].
log file: /home/dt/.ros/log/ea0002dc-9b51-11ed-997c-91c581871bdd/tagslam-1*.log
````

When I increase the   minimum_viewing_angle parameter this error goes away, however I'm looking for a better way to optimise and find these parameters. Any insights and help is appreciated, cheers.

dt1729, 2023-01-23:
My tagslam.yaml file: 

```
tagslam_parameters:
  optimizer_mode: fast
  minimum_viewing_angle: 35.0
  max_num_incremental_opt: 100
  pixel_noise: 2.0
  max_subgraph_error: 50.0
  subgraph_abs_prior_position_noise: 0.1
  subgraph_abs_prior_rotation_noise: 0.1
  use_approximate_sync: true
body_defaults:
  position_noise: 0.05
  rotation_noise: 0.01
default_body: perch
bodies:
 - perch:
    is_static: true
    default_tag_size: 1.0
    pose:
      position:
        x: -1.05
        y: 11.66
        z: 2.5
      rotation:
        x: 0
        y: 0
        z: 0
      position_noise:
        x: 0.0001
        y: 0.0001
        z: 0.0001
      rotation_noise:
        x: 1.0e-05
        y: 1.0e-05
        z: 1.0e-05
    type: simple
    tags:
      - id: 21
        size: 1.0
        pose:
          position:
            x: 0
            y: 0
            z: 0
          rotation:
            x: 1.57
            y: 0
            z: 0
          position_noise:
            x: 0.0001
            y: 0.0001
            z: 0.0001
          rotation_noise:
            x: 1.0e-05
            y: 1.0e-05
            z: 1.0e-05
      - id: 20
        size: 1.0
        pose:
          position:
            x: -2.35
            y: -1.5
            z: 1.5
          rotation:
            x: 1.57
            y: 0
            z: 0
          position_noise:
            x: 0.0001
            y: 0.0001
            z: 0.0001
          rotation_noise:
            x: 1.0e-05
            y: 1.0e-05
            z: 1.0e-05
 - rig:
     is_static: false
     type: simple
     odom_topic: "/odom"
     odom_frame_id: base_footprint
    #  T_body_odom:
    #   position:
    #     x:  -1.05
    #     y:  11.66
    #     z:  2.5
    #   rotation:
    #     x:  0.0
    #     y:  0.0
    #     z:  0.0
```

berndpfrommer, 2023-01-23:
That's a new one to me. Could you please post the whole console log that tagslam is spitting out (not just the error)? I suspect something like duplicate tags.

dt1729, 2023-01-23:
Yes, so i was trying to find the same, however tagslam-1.log doesn't exist even though there should be one.

berndpfrommer, 2023-01-23:
Sorry I really can't say what's going on there. I suspect you discovered a bug that bites when tags are dropped due to low viewing angle. This is something that's supposed to kick in only very rarely, but in your case it's happening all the time. This may trigger a bug that is otherwise undetected.
Can you create a rosbag that's small enough to be downloaded? I'd also need the exact tagslam.yaml, cameras.yaml, and camera_poses.yaml to reproduce the problem.

berndpfrommer, 2023-01-23:
Somehow the problem is not constrained properly. TagSLAM bombs out the first time it sees a tag with known pose (i.e. tag 20). But since your tagslam.yaml file looks good I suspect it has to do with the drops.

dt1729, 2023-01-24:
That sounds good to me, please have a look at these files that I'm using with the code: 

[https://drive.google.com/drive/folders/1pvBHGp265mASetoFkR77UB-Kbqr0yYmY?usp=sharing](https://drive.google.com/drive/folders/1pvBHGp265mASetoFkR77UB-Kbqr0yYmY?usp=sharing)

The bag file is small tests_gazebo goes with other test files and test.launch goes inside launch folder in the tagslam_tests folder.

Also, I wanted to ask other questions about Tagslam and making the development better and more understandable if someone wants to use some specific functions for the code. Can I mail you regarding the same?

berndpfrommer, 2023-01-24:
I had a look at your files:
- your tagslam.yaml has tag size = 1.0. So you are using 1m tags? Please check. To get any thing sensible I had to position your root tag (id 20) at a height of 5m. 
- your cameras.yaml has the intrinsics wrong. It should read like this:
  intrinsics: [1206.8897719532354, 1206.8897719532354, 960.5, 540.5]
- you are feeding in odometry. That's great, but odometry confers scale. So if your tag size is wrong, TagSLAM gets conflicting data from odom and tags.
- If you use odometry, you need to correctly specify the transform between odometry and body. You tried first with some absurdly large values (y: 11.66, that's meters!) and no rotations (fat chance your camera and odometry body pose are identical!). This way the odometry does more harm than good.
- Given the placement and viewing angles, you cannot run on tags only. You need odom to get decent results. But for odom to work, you need to work out the transform (most importantly, the rotation!) between your camera coordinate system (which is identical to your rig body) and whatever your odometry subsystem considers its reference frame.
- Below is the best I could get using fake_odom (note: you will get way better results using your odom, *but with the correct transform*)
[image]

dt1729, 2023-01-24:
Hello, I made the changes(or close) to what you suggested, but I'm still not able to get the results that you have shown in the figure, can you share your files with me so that I can recreate these results?

berndpfrommer, 2023-01-25:
1. Your pose was totally screwed up. I assume the robot moves in the ground plane. Your robot was not. To make it move in a ground plane I had to place tag 20 at about 5m high.
2. You can read about how to set the odom transform up here: https://berndpfrommer.github.io/tagslam_web/input_files/
3. The format in which the intrinsic calib is to be given can be found here (also read NOTE at the very bottom of page): https://berndpfrommer.github.io/tagslam_web/intrinsic_calibration/  

@dt1729 you can read about fake odom here:
https://berndpfrommer.github.io/tagslam_web/input_files/

Sorry the rviz graph I sent you was an intermediate result. I kept hacking the files trying to use your odom so I lost the exact files that produced the graph above (they were obtained by disabling your odom).

berndpfrommer, 2024-03-30:
closing this as resolved. Please re-open if there are still problems.

closed 2024-03-30 by berndpfrommer