berndpfrommer/tagslam #34: ROS2 Support

Alex-Beh, 2024-01-22:
Any plan for ROS2 support? Thanks

slmMuc2fly, 2024-03-06:
Great I love slamming

alvodo, 2024-11-29:
@berndpfrommer What is the current state of the ros2 branch? The last commit says that most is commented out, but when I tested it, it worked quite well, with the exception of a gtsam error once in a while.

berndpfrommer, 2024-11-29:
@alvodo I got a bunch of stuff working but I did not get around to testing it to the extent that I feel comfortable supporting it. Documentation is also missing.

alvodo, 2024-11-29:
@berndpfrommer I will keep you updated on any problems I run into. Is there anything simple and specific I should keep my eye on?

berndpfrommer, 2024-12-02:
I don't know of open bugs. But you should not get GTSAM errors. Please submit an issue if you have a simple reproducible case.

slmMuc2fly, 2024-12-02:
Is for me this message? Because I’m not sure how to use this a New technology. It is to confuse Daniel Ramirez +49 151 11534546Pestalozzistr.23a80468 MunichGermany

berndpfrommer, 2026-01-08:
switched ROS2 as the default branch. Still no documentation, but ROS2 is useful to some extent

closed 2026-01-08 by berndpfrommer