berndpfrommer/tagslam #37: How to convert mp4 shot by myself into bag (same as example.bag)

safsefefadsaasd, 2025-03-19:
Dear Bernd,  
I hope this email finds you well. I’m trying to create a workflow similar to your `example.bag`, but the generated bag file consistently differs from yours. Below is my process for your reference:  
1. Preprocessing:
I captured a 24 FPS MP4 video with a camera.  
Ran `video_publisher.py` to preprocess the MP4 file, generating a bag file named `video2bag_time.bag`.  
2. Launching Nodes:
 Opened `roscore` in one terminal.  
In a separate terminal, navigated to the `apriltag_ros` directory and executed:  
roslaunch apriltag_ros apriltag_detector_node.launch camera:=camera  
3. Recording Topics:
Opened a new terminal and ran:  
rosbag record -O new_output.bag /pg_17274483/image_raw/compressed /detector/tags  
4. Generating the Final Bag:
In another terminal, played the preprocessed bag:  
rosbag play --clock video2bag_time.bag  
This generated `new_output.bag` containing both `/detector/tags` and `/pg_17274483/image_raw/compressed`.  
Observed Issues:
While `rosbag info new_output.bag` matches your `example.bag` structurally, running the following commands produces unexpected results:  
  rosparam set use_sim_time true  
  roslaunch tagslam tagslam.launch run_online:=true  
  roslaunch tagslam apriltag_detector_node.launch  
  rviz -d `rospack find tagslam`/my_example/tagslam_example.rviz &  
  rosbag play --clock `rospack find tagslam`/my_example/new_output.bag --topics /pg_17274483/image_raw/compressed /detector/tags  

[image]

[image]

[image]

https://github.com/user-attachments/assets/827262da-3b27-429a-9902-dd8d811706b5

  - Only a static `tag_0` is displayed (even though tags 1, 2, 4, and 6 are visible in the video).  
  - TF warnings appear during execution.  
  - `new_output.bag` lacks `subgraph err` calculations and has misaligned timestamps (visible via `rqt_bag new_output.bag`).  
Additional Notes:
I directly copied your configuration files (`cameras.yaml`, `camera_poses.yaml`, `tagslam_example.rviz`, `tagslam.yaml`) from the example directory. Could this be causing issues?  
Would you be willing to review my `new_output.bag` for correctness? I’ve attached the following files for your reference:  
video_publisher.py  
test.mp4
new_output.bag(can be made by video_publisher.py and test.mp4)
Thank you for your time and assistance!

video_publisher.py :
# -*- coding: utf-8 -*-
import rosbag
import cv2
from sensor_msgs.msg import CompressedImage
from cv_bridge import CvBridge
import rospy
 
import numpy as np
 
def video_to_bag(video_path, output_bag):
    bag = rosbag.Bag(output_bag, 'w')
    cap = cv2.VideoCapture(video_path)
    cb = CvBridge()
    
    prop_fps = cap.get(cv2.CAP_PROP_FPS)
    if prop_fps != prop_fps or prop_fps <= 1e-2:
        print("Warning: can't get FPS. Assuming 60.")
        prop_fps = 24
    prop_fps = 24
    
    ret = True
    frame_id = 0
    while ret:
        ret, frame = cap.read()
        if not ret:
            break
        stamp = rospy.rostime.Time.from_sec(float(frame_id) / prop_fps)
        frame_id += 1

         # 逆时针时针旋转90度*0
        rotated_frame = np.rot90(frame, k=0)

        compressed_image = CompressedImage()
        compressed_image.header.stamp = stamp + rospy.Duration.from_sec(1701295483.21)
        compressed_image.header.frame_id = "pg_17274483"
        compressed_image.format = "png"  # 使用PNG压缩格式
        
        # 将图像转换为压缩格式
        _, compressed_data = cv2.imencode('.png', rotated_frame, [int(cv2.IMWRITE_JPEG_QUALITY), 90])
        compressed_image.data = compressed_data.tostring()
        
        bag.write("/pg_17274483/image_raw/compressed", compressed_image, compressed_image.header.stamp)
 
    bag.close()
    cap.release()
 
# 输入视频文件和输出bag文件名
video_path = '/home/a/catkin_ws/src/tagslam_test/tests/test_25/test.mp4'
output_bag = 'video2bag_timeX.bag'
 
# 调用函数进行转换
video_to_bag(video_path, output_bag)

berndpfrommer, 2025-03-21:
I tried to follow your steps but already this step fails:
```
roslaunch apriltag_ros apriltag_detector_node.launch camera:=camera
```
because there is no  ``apriltag_detector_node.launch`` file.

Use this command to produce a bag with the detected tags:
```
roslaunch tagslam sync_and_detect.launch bag:=`pwd`/video2bag_timeX.bag
```

But that's not the only problem. Once you have done that, you need to produce a matching tagslam.yaml and cameras.yaml file. You can't just take the example files because the tag sizes may be different from what's specified in tagslam.yaml, and your camera calibration (cameras.yaml) is totally different.

Please read up on camera calibration and how to produce a valid tagslam.yaml file.

safsefefadsaasd, 2025-03-22:
Sorry, I use this command to produce a bag with the detected tags: video2bag_timeX.bag_output.bag
roslaunch tagslam sync_and_detect.launch bag:=/home/a/catkin_ws/src/tagslam_root/src/apriltag/apriltag_ros/video2bag_timeX.bag
But when i use this command to check (tagslam.launch's internal related bag names and corresponding changes)
roslaunch tagslam tagslam.launch bag:=`rospack find tagslam`/my_example/video2bag_timeX.bag_output.bag

[image]

It appears that err is still always equal to 0, Indicates that the factor graph is not optimized?
But when I use this command to check your example.bag and example.bag_output.bag, there err is not equal to 0

[image]

can you tell me what's wrong happened,is it because my mp4 or the original bag video2bag_timeX.bag wrong?
If it is convenient for you, can you convert my mp4 into video2bag_timeX.bag through my video_publisher.py, and then generate video2bag_timeX.bag_output.bag through the instructions you gave me. And then run roslaunch tagslam sync_and_detect.launch bag:=`pwd`/video2bag_timeX.bag to generate video2bag_timeX.bag_output.bag and check its err?