berndpfrommer/tagslam #38: subgraph err is always equal to 0

safsefefadsaasd, 2025-03-22:
Sorry, I am the person who asked questions two days ago, and the information involved in this question is included in the last question.
I use the command you give me yesterday to produce a bag with the detected tags: video2bag_timeX.bag_output.bag
roslaunch tagslam sync_and_detect.launch bag:=/home/a/catkin_ws/src/tagslam_root/src/apriltag/apriltag_ros/video2bag_timeX.bag
But when i use this command to check (tagslam.launch's internal related bag names and corresponding changes)
roslaunch tagslam tagslam.launch bag:=`rospack find tagslam`/my_example/video2bag_timeX.bag_output.bag

[image]

It appears that err is still always equal to 0, Indicates that the factor graph is not optimized?
But when I use this command to check your example.bag and example.bag_output.bag, there err is not equal to 0

[image]

can you tell me what's wrong happened,is it because my mp4 or the original bag video2bag_timeX.bag wrong?
If it is convenient for you, can you convert my mp4 into video2bag_timeX.bag through my video_publisher.py, and then generate video2bag_timeX.bag_output.bag through the instructions you gave me. And then run roslaunch tagslam sync_and_detect.launch bag:=`pwd`/video2bag_timeX.bag to generate video2bag_timeX.bag_output.bag and check its err?

safsefefadsaasd, 2025-03-22:
Due to potential time zone differences, I am consolidating my questions within this issue to expedite the resolution process. I appreciate your patience and guidance:

1、Configuration File Confusion:
    I have been comparing my bag files with the example files (example.bag and example.bag_output.bag). I understand that the critical configuration files to adapt are cameras.yaml, camera_poses.yaml, tagslam.yaml, and tagslam.launch. However, even after replacing these files with my own versions, the system still runs without errors. This is puzzling—are there additional files or parameters I might be overlooking?

2、MP4-to-BAG Conversion Issue:
    My primary challenge lies in converting my recorded MP4 videos into the example.bag format. Despite following the workflow described in my previous closed issue and this one, I suspect flaws in my process. If you could replicate the steps outlined in these issues to verify the workflow, I would greatly appreciate it.

3、Discrepancy in Tag Display and Optimization:
    In your example tagslam.yaml, only tag_0 is explicitly defined, yet RViz displays multiple tags (e.g., tag_1, tag_2, etc.), and the factor graph optimization completes successfully. In contrast, my setup (screenshots attached in this issue) fails to achieve this. Could you clarify why your example works seamlessly while mine encounters issues?

Your insights on these points would be invaluable. Thank you for your time and support!

berndpfrommer, 2025-03-22:
Please post your tagslam.yaml and cameras.yaml.
Also please describe how you got your camera calibration.

safsefefadsaasd, 2025-03-22:
camera.yaml:
cam0:
  camera_model: pinhole
  intrinsics: [1404.917, 1403.505, 962.317, 605.784]
  distortion_model: radtan
  resolution: [640, 480]
  rostopic: /pg_17274483/image_raw
  tagtopic: /detector/tags
  rig_body: rig
(I only know my camera's resolution and know that it don't have distortion_coeffs, other's copy your example's cameras.yaml)
However,I'm confused how to design tagslam.yaml (these confuses are in this issue). Could you help me to design my tagslam.yaml by the mp4 and other data I give you in issue #37. If it is feasible,could you visualiz my bag in rviz. Sorry to bother you！

safsefefadsaasd, 2025-03-22:
What's more, when i rosbag play my bag and rostopic echo /detector/tags, i find that the seq is always equal to 0, 
(I use this command to produce a bag with the detected tags: roslaunch tagslam sync_and_detect.launch bag:=`pwd`/video2bag_timeX.bag)
[I'll wake up tomorrow morning and check my messages immediately! Have a good day! ]
[image]

berndpfrommer, 2025-03-22:
OK, so from what I gather you don't have a valid camera calibration. You *cannot* just take the calibration data (or parts of it!) from the example files.

Please perform a camera calibration first following these steps, then report back with a valid calibration file.
https://docs.ros.org/en/rolling/p/camera_calibration/doc/index.html

Until you have a valid camera calibration there is no point in me trying to run anything.
The only other thing I need to know is the size of your tags, i.e. the edge length of a tag.

safsefefadsaasd, 2025-03-23:
Thanks! I have run it, but there are still some problems, I have uploaded my case to the web disk. 
https://pan.baidu.com/s/1edqMEaNIIc_KkaQs4z2AOg    Extract code:  unvr
The instructions to run are: (on five different terminals, all under catkin_ws)
rosparam set use_sim_time true
roslaunch tagslam my_example_tagslam.launch bag:=`rospack find tagslam`/my_example/my_example.bag run_online:=true
roslaunch tagslam apriltag_detector_node.launch
rviz -d `rospack find tagslam`/my_example/tagslam_example.rviz &
rosbag play --clock `rospack find tagslam`/my_example/my_example.bag --topics /pg_17274483/image_raw/compressed /detector/tags

The results is also under my_example, with several problems:
1、I changed the format of my_example.bag to be exactly the same as your example.bag, including the same topics, but it doesn't show the image
2、There are four tags: tag_1, tag_2, tag_4, and tag_6
But in rviz, you can see that tag_1 and cam overlap, which makes me very confused.
Can you run my_example to see, really trouble you!
tips: Maybe because of the selection of initial values, the optimization effect of factor graph is not good.
The length and width of the tag are 0.16m

berndpfrommer, 2025-03-23:
I cannot access your web disk. Please post the results of your calibration in clear text here on GitHub. Please post the original file as produced by the ros calibration software. Also post the console output of your calibration so I can see what the reprojection error is.
When you post on github, avoid posting screenshots. Instead, cut and paste console output, and enclose it with "code" markers in markdown format.

safsefefadsaasd, 2025-03-23:
here you go:
https://github.com/safsefefadsaasd/tagslam_mybag

berndpfrommer, 2025-03-23:
The calibration you posted is not from the camera with which the mp4 file was generated because the resolution of the cameras is different.

I'm not sure you really exist, or if you are some AI bot. Your github profile is blank. Your posts are very verbose and almost make sense, but not quite.

Sorry if you are a real person, but I'll close this issue now and ask you to not open new ones or else I will block you. Please look for help elsewhere.

closed 2025-03-23 by berndpfrommer